icebreaker-fpga/icebreaker #14: Explore the possibility of using a low cost STM32 instead of FTDI

esden, 2018-09-10:
There are very low cost STM32 that don't even require an external oscillator to be used with USB.

STM32F042K6 for example can be purchased on aliexpress for ~$1.50. I have also compared prices of similar parts from other vendors. The STM32 turns out to be lower cost than any alternative chip with this kind of specifications. With added bonus of being easily available from a lot of vendors on aliexpress.

List of advantages of the STM32:
* significantly lower cost: ~$15 lower retail cost of the icebreaker
* less parts and placements: no external flash chip and no external oscillator needed
* automatic unique serial number in the descriptor strings generated from the unique id of the STM32. No need to track unique serial numbers in manufacturing.
* flexible firmware on the STM32 alowing for updates and new features in the future
* Possible to wire up so that we don't need to use any jumpers to choose FLASH or FPGA flashing

List of disadvantages of the STM32:
* Need to develop firmware for the STM32
* Need to develop desktop program for flashing (multiplatform software developemnt)
* Mess with drivers depending on which model we use (windows headaches mostly)
* No high speed USB just full speed.
* No optional asynchronous FIFO mode.

List of things we retain:
* Ability to flash the on board FLASH chip
* Ability to upload bitstream to the FPGA directly
* Ability to connect to the UART serial port (USB to serial converter)
* Generate 12MHz external clock for the FPGA (STM32 MCO pin)

This issue is here to track the testing of that part. Close it if we decide not to go down that path, or we realize that it has some big flaw that is worth the 4x price difference of the FTDI chip. Or when we decide to use this approach and decide that this is our main path.

esden, 2018-09-10:
Added a list of alternative chips to the STM32.

I am still planing to continue down the road with STM32 as I know that chip family very well. They are abundant and easy to get from many sources. The price is comparable with the other chips. And I have many projects to source code from. Also STM32 is best supported by Black Magic Probe and I like to be able to be able to debug the things I build. :)

esden, 2018-09-10:
After a discussion with @cliffordwolf here is a collection of thoughts about how we want to build the USB interface on the device:

Requirements:
* Has to be easy to setup on all the major three OS, with no driver installation if possible:
  * Linux
  * Windows
  * MacOS X
* Has to have a simple protocol but preferably not bitbang as it is super inefficient
* Provide FLASH programming (SPI)
* Provide USB-Serial for simple FPGA designs (UART) (this is much simpler than adding a full USB stack to the design that is why an external chip is being considered at all)
* The programming protocol has to support the following features:
  * Flash the SPI FLASH containing the FPGA bitstream
  * Upload bitstream directly to the FPGA SRAM cells
  * Reset FPGA
  * Reset FLASH and get it out of QSPI mode for flashing

Resulting design decisions:
* Due to how ugly Windows driver situation is we prefer to use a builtin virtual serial driver instead of going raw libusb protocol.
* The device presents the following hybrid USB interface to the PC:
  * CDC-ACM (Virtual Serial) port No. 1 for FPGA flashing
  * CDC-ACM (Virtual Serial) port No. 2 for USB-Serial adapter interface
  * DFU stub, gives access to the DFU bootloader of the STM32 for programmer firmware upgrade.
* Base the programming protocol on the FTDI MPSSE engine opcodes. This way adding support for this programmer to iceprog will be relatively simple. How viable this is, depends on how much extra functionality we will want to add that is currently not possible with the FTDI programmer solution.

mithro, 2018-09-10:
FYI - You can perfectly emulate a FTDI chip with these low cost STM32 chips
(just a matter of software :-)

esden, 2018-09-10:
@mithro I bet this is possible. I am not sure what benefit it provides besides the direct reuse of existing programmer desktop software. I do like the idea, it might be either the default or an alternate use of the stm32 programmer.

What I like even more is to be able to do things that the FTDI chip does not provide. For example being able to program the flash chip or the FPGA sram directly without having to set jumpers. Having a feature like this would somewhat offset the loss of high speed USB and FIFO modes.

mithro, 2018-09-11:
@esden Emulating the FTDI chip is kind of silly -- except it stops people complaining about you not using the FTDI chip.....

ekawahyu, 2018-10-02:
@esden what would be the USB library you have in mind? I did some code with STM32 library for CDC-ACM and DFU capability on F072. It was like 48kB+, so probably it won't fit in F042K6T6. It was compiled with GCC, it would probably much less with paid compilers.

gsteiert, 2018-10-03:
How about MSC with UF2 instead of DFU?
There is an STM port:
https://github.com/mmoskal/uf2-stm32f
It really simplifies the windows driver issues and you don't have compile a programming app for every OS.

mithro, 2018-10-03:
I agree that having support for UF2 would be excellent.

The great thing about having an ARM micro is that you could potentially support *all* these modes. UF2, DFU, FTDI serial, etc.... It's just different software code running on the microprocessor.

UweBonnes, 2018-11-08:
The FT2232H allows to pump data to the PC with up to 30 MByte/s. Some Full-Speed device like the STM32F042 can only do up to 1.5 MByte/sec...

osterwood, 2018-11-08:
The ATSAMG55 looks promising.  The G package gets down to $2.92 in QTY 1k when purchased direct from Microchip.  Impressive specifications and have USB HS.  SKUs are:

* ATSAMG55J19A
* ATSAMG55J19B
* ATSAMG55G19A
* ATSAMG55G19B

G are 49/WLCSP package and J are 64/LQFP (they are a little more expensive).  I've never used them before, but might be worth considering.

esden, 2018-11-08:
@osterwood are you sure they have USB HS? From the datasheets it seems they only have a FullSpeed transceiver built in. But I might be missing something. There are plenty of ARM Cortex chips that have HighSpeed support but require an external PHY. I was hoping to avoid that.

UweBonnes, 2018-11-08:
STM32F723 has HS Phy on chip, but pricey also...

osterwood, 2018-11-08:
Yeah - you're right.  Saw "high speed" on the summary page and just assumed.  Looks like the only options from Microchip are PIC32MZ ($6 to $7) and SAMS7x/E7x/V7x ($5 and up)

mithro, 2018-11-08:
If you want full 480 megabit, the FX2 chip is a decent option and can get down to $3-4 USD each. They are used on our Opsis and Atlys board.

@whitequark has been working on doing FX2 firmware to support UF2, Serial and DFU programming modes for me.

UweBonnes, 2018-11-09:
The STM32F730Z8T6 (with integrated OTG HS phy) comes need to the 3$@1000 margin.

smunaut, 2018-11-09:
Also need to check that whatever uC with HS you use, it not only has high-speed usb, but also some other peripheral to actually transfer all that data to the ice40 and good DMA between usb and it.

UweBonnes, 2018-11-10:
It is a pity that the smaller STM32F72x packages don't have the USB-HS phy pinned out.

gsteiert, 2018-11-12:
Has anyone ever used a Nuvoton mcu?
They have cost effective high speed options in a smaller package:
https://direct.nuvoton.com/en/m484sidae

esden, 2018-11-12:
@gsteiert I personally never did, but I know that people like using nuvoton parts in toys as they are usually very low cost. Good find! :)

esden, 2018-11-13:
@gsteiert that is a really awesome find! It pretty much perfectly matches our needs.

The next question is how good is the documentation of the chip, and if there are existing open-source friendly libraries. If there are no libraries it is not a huge problem but rather a delay as we need to create them, as long as the chip is properly documented. ;) Do you have any experience developing for Nuvoton chips using GCC/LLVM/GDB/Make and all the other typical open source embedded tools? And no Keil or IAR is not an option for the project beyond very basic evaluation tests. ;)

I briefly looked at the [reference manual](http://www.nuvoton.com/resource-files/TRM_NUC505_Series_EN_Rev1.07.pdf), and it seems to at least contain register definitions and it can be downloaded without much trouble. As long as the chip is well documented I think we might have a winner! :D

gsteiert, 2018-11-13:
It doesn't include the NUC505 specifically, but several of their other parts are supported in mbed:
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NUVOTON
This is apache 2 licensed and you can build it with gcc.  Even if you don't want to use their HAL, it is a good source for the CMSIS libraries.

xelll, 2018-11-15:
I look forward to your success

majbthrd, 2018-12-02:
The ATSAMD11 is only $1.24 in single quantity and $1.02/3k.  I have an open-source DFU protocol bootloader (https://github.com/majbthrd/SAMD11-USB-DFU-Bootloader/) that is only 1kByte in size.  Providing a CDC implementation and FPGA initialization in 15kBytes is hardly challenging, and I'm familiar with the iCE40.  If your collective hearts are really set on the STM32, I also have example CDC and composite code for the STM32F042/STM32F072 on my github page.

smunaut, 2018-12-02:
The SAMD11 is Full Speed only, not High-Speed.

majbthrd, 2018-12-02:
I am aware the SAMD11 is FS, thanks.  An iCE40UP5K configuration consists of 104,158 bytes.  The perceptual difference between a FS and HS for such a download is negligible, particularly if SPI erase and write operations are involved.  There would be a difference for writing the entire 16MByte quad flash, but you guys have ask yourselves whether this board is something with good value for money for education or something else.

majbthrd, 2018-12-02:
Good luck with your project.  I have designed in both STM32F0x2 and SAMDx1 (both sub-$1 in quantity... the figure I quoted was a snapshot from a US distributor and likely reflects trade tariffs), and both have their particular strengths and weaknesses.  About the only thing I would politely disagree with is an assertion earlier in the thread that CDC would take 48kBytes+ in the STM32.

esden, 2018-12-02:
@majbthrd thank you. Yes you are probably right that CDC can be put in a much smaller Flash space. It was not my opinion that we can't put that into a tiny chip. Especially when you don't use the vendor libraries that are made to be very generic and tend to be very inefficient. The chip used at the end just has to fit a certain price and peripheral requirement list. (I should update my top post to correct the goals) Whatever fits the bill best is what we will try going for. The Nuvoton part seems to fit that bill best at the moment, although I am very concerned with their sparse documentation. We will see how that goes. :)

majbthrd, 2018-12-13:
I acquired a NuTiny-NUC505Y eval board and set about getting USB working on it.  It is extremely far from a finished solution, but I've posted a snapshot of a CDC virtual serial port implementation here: https://github.com/majbthrd/NUC505testing  Throughput in Linux using dd is about ~9.5+ MBytes/s in each direction (e.g. ~152+ Mbits/s total).  Code size is under 9kBytes.

esden, 2018-12-14:
That is awesome! Thank you for sharing! That is definitely a great starting point! :D

kehribar, 2019-01-08:
Hi All,

That Nuvoton chip sounds really promising. 

Anyway, I just wanted to drop my ongoing work with STM32F042 here. May be useful to someone. My goal is to create some sort of low cost "ICE40 programming / basic computer interface" platform with minimal FPGA IO consumption. For _real_ data transfer needs (like raw ADC capture etc.) user can connect an additional FT232H single channel 480Mbps fifo to the GPIOs of FPGA.

Repo link is this: https://github.com/kehribar/ice40helper

Right now the [firmware](https://github.com/kehribar/ice40helper/tree/master/firmware) (libopencm3 based) & [software](https://github.com/kehribar/ice40helper/tree/master/software) can:

* Read external Flash content when boot up and program the UP5K over SPI. 
* Program the UP5K (non volatilely via SPI) _on demand_ via USB.
* Program the SPI Flash attached to MCU over USB.
* Can do basic SPI data [read / write exchange](https://github.com/kehribar/ice40helper/tree/master/fpga/up5k/spi_rw) between MCU and the FPGA. I'm planning to work more on this to explore the data transfer limits.
* F042 has USB ROM bootloader and it doesn't require any external programmer for firmware experimentation.

Beware:

* USB protocol is CDC based which means it appears just like a regular serial port in the computer side and doesn't need any low level libusb etc. programming in the host side. But, it doesn't do _transparent_ USB <--> SPI bridge. Everything is command transmit / reply based. 
* SPI Flash is _only_ connected to the MCU. (Just like in whitequark's Glasgow project) FPGA can't reach its content directly. This may be the only hardware caveat.
* Most of the things in the firmware (like SPI clock frequency, packet length etc.) are hardcoded and non documented. There is a lot of room for improvements for sure. 

You can check my reference hardware design from here: [schematic pdf](https://github.com/kehribar/ice40helper/blob/master/hardware/up5k_v0/up5k_v0.pdf)

zeroeth, 2019-09-22:
A little late to the party, but just want to +1 stm32 because of all the great libopencm3 examples/workflows.

Having the board potentially be dual purpose (and custom usb hid such as usb joystick/midi) would be absolutely jawesome.

related: dadamachines did something similar so the board is directly arduino compatible too https://dadamachines.com/product/doppler/ some food for thought! Especially now that arduino themselves have a mostly closed toolchain fpga hybrid board.

ejhollas, 2019-10-12:
Windows drivers can be simplified by making use of Microsoft's WinUSB drivers. These drivers are built-in to Windows 8 and later. A simple INF file can make it work on Windows 7 as well. 

To get Windows to attach WinUSB, the ST firmware will just need a few USB descriptors that Windows will read on first plug-in. No drivers need signing using this method. WinUSB supports composite devices, so we can expose a CDC serial interface and a vendor specific interface. This will allow us to use high bandwidth endpoints over bulk or iso.

The WinUSB API has native C bindings. There is support in the libusb project as well as C#, go and python bindings. I'll be happy to help contribute and test the descriptors.

rljacobson, 2019-11-28:
I am curious to know if V2.0 is being actively developed, as there does not seem to be much activity on this issue. @majbthrd's experimentation seems to indicate that the Nuvoton chip is in some ways quirky but serviceable. The STM32F730Z8T6 has the advantage of familiarity, runs at twice the clock rate, but costs slightly more and has comparatively little FLASH ("external memory can be used with no performance penalty thanks to the L1 cache").

gsteiert, 2019-11-28:
I think there are a few new options since this thread was started.  The
i.MX RT1064 used on the Teensy 4.0 is even faster, or the RT1010 for a
lower cost option.  The LPC55xx series has integrated flash.  All of these
have the high speed USB phy integrated.

UweBonnes, 2019-11-29:
Greg Steiert writes:
> I think there are a few new options since this thread was started. The
> i.MX RT1064 used on the Teensy 4.0 is even faster, or the RT1010 for a
> lower cost option. The LPC55xx series has integrated flash. All of these
> have the high speed USB phy integrated.
> 
I see one problem in a missing USB HS stack that is open source and
works with a makefile

-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 1623569 ------- Fax. 06151 1623305 ---------

tannewt, 2019-11-29:
I'd like to add high speed support to [TinyUSB](https://github.com/hathach/tinyusb) (an open source USB stack). @hathach just added full speed support for the i.MX RT series.

hathach, 2019-11-29:
@tannewt thanks, iMX RT series is actually  highspeed USB :)

ulixxe, 2021-11-06:
Another possibility is to implement USB device stack on the same FPGA fabric. See:
https://github.com/ulixxe/usb_cdc

smunaut, 2021-11-06:
Did you even _read_ the thread ?

gregdavill, 2022-02-12:
There are now WCH parts available* that integrate a USB HS PHY. Like Gigadevices they have both CorteX-M3/RISCV variants, they also have some "RISC" based devices for implementing interface controllers, these appear to fill a similar market to the FX2 series, not sure if they're also 

Here are some RISCV based options: 
- CH565/CH569, USB2/3 with built in PHY. (Likely extra cost since you're paying for the USB3 PHY, when it's not needed)
- CH563, USB2 HS with built in PHY.
- CH32V305/307, USB2 HS with built in PHY.

Cortex M3 options:
- CH32F205/207, USB2 HS with built in PHY.

All of these also include USART+SPI so should satisfy the base requirements for interface to the FPGA.

LCSC has stock of the CH32V307, other parts I imagine you'll have to contact WCH for a sales channel, like Gigadevices.

The main downside these parts have is their size, because they also include a fair few extra interfaces the smallest parts are in LQFP-100/LQFP-64. So might end up making the board a bit dense.

psherman42, 2025-02-15:
Would really like to see a small basic board without any USB whatsoever, regardless FTDI or otherwise. Leave the host interface as an optional attachment to be decided by the user. In other words, modularize rather than integrate. There may be people out there, now or in future, wishing for a retro UART or RS-232 host link ...

fanoush, 2025-02-17:
> The main downside these parts have is their size, because they also include a fair few extra interfaces

I know this is old comment but recently I got some cheap CH32V305 based debug dongles and they work quite fine with HS mode when reflashed to  https://github.com/prosper00/CH32V305-DAPLink-HS  and the TSSOP20 part used there is not big.

The dongle is either WCH-Link E (about $5 on aliexpress) or there is mini version 
[image] for about $2.50