icebreaker-fpga/icebreaker #15: Add Access to FT2232H VREGOUT +1V8 for Powering VCCIO2

RossBencina, 2018-09-22:
Greetings folks, I would like to make a suggestion. I can turn this into a PR if there is interest.

The FT2232H has as regulated 1.8V output. VCCIO2 provides a jumper for supplying a different IO voltage. Wouldn't it be nice if +1V8 could be jumpered to VCCIO2? For example there are a number of single board computers with 1.8V GPIO and I2S, it would be nice to be able to connect the icebreaker PMOD to them directly.

I believe that powering VCCIO2 FT2232H VREGOUT off would be compatible with the UP5K power sequencing requirements.

I can see two ways to make this happen:

1. Provide a test pad to allow greenwiring from the the pad to the jumper (see below)
2. Route +1V8 through one of the middle layers and provide a 3-way jumper for VCCIO2. (Replace J17 with a 3-way jumper). I have not tried this but it looks possible.

To show what I mean, for option (1) here is how it could work:

Add test point:
[image]

On F.Cu, add via to route +1V8 to B.Cu:
[image]

Add pad to B.Cu:
[image]

esden, 2018-09-23:
That is great! You are the second person to bring up the 1v8 supply. I was about to add a way to expose that net. The only thing I need to check is what the allowed load on that regulator inside the FTDI chip is. Also it would be nice to find a way to bring that power net closer to the jumpers. If you already have your changes done feel free to open a pull request. It could be a good starting point for me.

Thank you very much for the idea and the time! :)

RossBencina, 2018-09-27:
@esden Sorry, only just found time to get back to this. I've made a pull request with my Test Point changes, but I don't recommend that you merge it. The diff to icebreaker.kicad_pcb seems very large (maybe because I shoved some traces to place the test point).

As I said above, I think the best approach is to make a 3-way jumper for VCCIO2 and route the 1v8 signal  on the (middle) power plane.

I too haven't researched whether the FTDI can source enough current for this to be safe. And if you switch to STM MCU maybe this whole thing becomes moot.

@alexhude: so far this update is for VCCIO2 only. (Which is all I need ;) ) But maybe you can make a case for adding it to other ports?

commit b096366c (2018-10-14): Initial commit of iCEBreaker V1.0b.

* Added 1V8 jumper for FPGA GPIO bank 2. Not added to 0&1 as the pins on
that bank are also connected to the FTDI chip. Extra precausions have to
be taken in that situation as the FTDI chip GPIO are connected to 3.3V.
The user will have to assure that there are no problems and the FPGA
will not fry either itself or the FTDI chip. We do not want to encourage
that, but if you feel adventerous you are welcome to try. ;) Closes #15
closed 2018-10-14 by esden