icebreaker-fpga/icebreaker #18: Add another CS line from FTDI to the iCE40 to use SPI comm from host -> fpga

smunaut, 2018-11-04:
Having SPI communication from host to the ice40 has many benefits : 
 - Very simple protocol to implement (I'd say even simpler than UART since it's synchronous)
 - Can be clocked fast (Up to 30 MHz)
 - Re-use most of the existing pins used for the configuration.

The only thing needed is a way to differentiate between host accessing the flash and the host accessing the user logic in the fpga.

This can be done one of two ways : 
* Using discrete logic gates. Since host access to the flash is only going to happen when CDONE is low (or CRESET is low), you can use single gate 74 logic to mask the CS signal to the flash.
However this solution at this stage is not ideal because :
   - Non trivial change to the design
   - Adds parts (and thus BOM cost and pick and place time and ...)
   - Interferes with access of fpga logic to the flash once booted.
* Use another distinct CS line from the FTDI to another iCE40 pin. Best candidates are the two leds signals LEDG / LEDR or the button signal. I like the LED signal because this way I still have one button and one led for user logic. And also the led blinks when there is SPI access from the host, providing a visual indicator of access.

So all in all, I think a solder jumper from LEDG to ADBUS3 would be a good idea.

esden, 2018-11-18:
I went with the ADBUS3 to LEGR solution. Cheers! :D

closed 2018-11-18 by esden

AnasMalas, 2025-01-09:
@esden 
To clarify - in V1.0e, the jumper is connected by default, so I should avoid driving LEDR from the FPGA to avoid damage - as this is an output pin on the FTDI chip. Is this correct? If so, I think this should be included in the readme

commit c7a39f94 (2018-11-18): Added jumper from ADBUS3 to ~LEDR. Closes #18

Thanks for the suggestion @smunaut :D