icebreaker-fpga/icebreaker #19: Add unpapulated QSPI RAM footprint

esden, 2018-11-05:
To add expanded RAM amount to the iCEBreaker without sacrificing the PMODs we should have an SO8 unpopulated/optional footprint for the LY68L6400.

We can think about a better setup for this in the V2.0 hardware cycle.

esden, 2018-11-18:
As per @smunaut suggestion. I have added a pad that allows for an easy addition of an SPI RAM chip in SO8 package piggybacked onto the existing flash chip. We can't use ram and flash at the same time but we can use them in the same design by using ~LEDG signal as CS for the RAM. It is good enough for V1.0 ;)

closed 2018-11-18 by esden

umarcor, 2020-12-17:
@esden, I can't identify that pad in the schematics or the figures in the readme. Was it removed in the end?

smunaut, 2020-12-17:
Next to pin 37 of the fpga in https://github.com/icebreaker-fpga/icebreaker/blob/master/hardware/v1.0e/icebreaker-sch.pdf

umarcor, 2020-12-17:
@smunaut thanks. I misunderstodd the meaning of "piggybacked" and I was looking for an unpopulated footprint at the bottom. All clear now.

esden, 2020-12-17:
@umarcor Right, it might not be clear for everyone. I am putting a photo of how it comes together for reference.

[image]

commit c828c31d (2018-11-18): Added easier way for adding "piggyback" SPI RAM.

Added a pad near the SPI flash CS pin. This pad is connected to ~LEDG.
This will allow us to solder on an SPI RAM chip on top of the SPI flash
chip and connect it's CS line to ~LEDG signal. Allowing us to use the
flash and RAM through the same SPI interface.

This closes #19 for the V1.0c milestone. We will revisit this topic for
V2.0 milestone.