icebreaker-fpga/icebreaker #23: PMOD connector 3D model error

DanSGiesbrecht, 2019-01-03:
Hi @esden 

I was combing through your design (v1.0c) and noticed that J21 has a duplicated 3D model:

[image]

esden, 2019-01-03:
The offset model should not show up. The second 3d model is a model from the older KiCad 3d library. You might have both installed. You might want to delete old 3d models and only have the new ones installed. I left the old models just in case and prefixed them with `dis` so that KiCad can't find it. It is strange that it shows up at all. What KiCad version do you have installed?

[image]

[image]

DanSGiesbrecht, 2019-01-03:
Thanks for the quick response! I'm glad to hear that it's just on my end. I have KiCad 5.0.1 installed, with the default 3D library.

esden, 2019-05-13:
Thank you very much for the contribution. This should now be fixed in all the versions of the board. :)

closed 2019-05-13 by esden

madding1602, 2023-06-14:
Hi. I'm trying to get the CAD file made for a project. I need the 2 PMOD connectors on the side, plus the RGBLED, FLASH and CRESET (straight male), along woth the 2 5V connectors to the 5V (straight females). How can I add them? I haven't used KiCAD, but when I try to create the STEP file it's created with no connectors (and I've tried multiple versions)

esden, 2023-06-14:
@madding1602 This is not the right place to ask the question. It is bad style to jump onto some other github issue that has nothing to do with your project. You should join the KiCad Discord server and ask there instead: https://discord.gg/xvnRc9EY

That said, if you want the part footprints to export together with the board using STEP, you have to make sure that the part models are also STEP and not WRL. If you have further questions please ask them on the KiCad User Discord server.

madding1602, 2023-06-14:
> @madding1602 This is not the right place to ask the question. It is bad style to jump onto some other github issue that has nothing to do with your project. You should join the KiCad Discord server [...]

okay. thanks!

commit 4e2853ec (2019-05-13): Fixed pin header 3d models.

We had some of the models doubled up or not fully disabled. Now the TH
pin headers are all using KiCad V5 3d model locations and are all
disabled.

This fixes #23