icebreaker-fpga/icebreaker #24: Add via in pad to the USB footprint to anchor the pads better to the PCB

esden, 2019-01-03:
The USB connector is pretty sturdy, but we can make it even better if we add via in pad to the ground pads to anchor the connector even better to the PCB.

cliffordwolf, 2019-01-03:
Haha. I already wanted to propose that but thought it would just show how little I know about design for manufacturing. :)

(Would have been expecting something like reduced yield due to soldering issues because the via is also a heat sink.)

esden, 2019-01-03:
Yes that is definitely a risk. Also the wicking of solder down the via can be an issue. But the pads are pretty large and generously covered with solder paste, and the improved anchor can be worth the risk. I will have to think about it a bit and maybe even make a test board with different footprints and make some force tests. :)

emeb, 2019-01-03:
To prevent solder wicking down the vias you could make the pads larger and put the vias underneath mask, with only the mask opening for the socket contacts per mfg specs.

smunaut, 2019-02-15:
Well I would think that getting the via filled with metal would only help the mechanical strength, but of course you need enough solder on there :p

esden, 2019-05-15:
I am postponing addressing this issue. I somehow doubt that just having VIA will actually increase the adhesion of the copper laminate by a significant amount.

I am testing out footprints with solder retention legs on a separate board. If that works out we will go that route. But that will not happen for V1.0d but 1.0e probably.

esden, 2019-05-15:
One more thing. I have connected the shield pads solidly to the rest of the copper pour. This increases the size of the copper laminate that the footprint is connected to. Remembering how the one board looked like that got it's connector ripped off, the failure point were the thermals connecting the pads to the pour. I hope this will improve the mechanical strength for now.

And as always: be nice to your hardware folks, and don't let it dangle under your feet, so you don't trip. :)

esden, 2021-12-18:
As V1.0e is already produced. And the next version of the board will likely have a USB-C connector. I think this issue is irrelevant by now. So I will close it.

closed 2021-12-18 by esden