icebreaker-fpga/icebreaker #25: Add Bill of Materials

stv0g, 2019-01-10:
I am currently, working on creating a BOM with MPN's and DigiKey part numbers.
Anyone wants to help?

esden, 2019-01-10:
Please do not do that. It is easy enough for an individual to put the board together. We don't need to make the work for cloners even easier. If you have questions about the Bill of Materials please ask me directly in Gitter, I will be happy to help you out: https://gitter.im/icebreaker-fpga/Lobby

closed 2019-01-10 by esden

esden, 2019-01-11:
If you want to DigiKey part list I might be able to refer you to someone who might already have one. Will save you time. It might be slightly convenient to have this publicly available, but I am usually available on Gitter and will try to help out anyone who asks in person for their personal use. I hope you understand. :)