icebreaker-fpga/icebreaker #27: Sync FIFO mode support

smunaut, 2019-04-10:
The current version only allows the async fifo mode wit the ftdi to be used. But the async mode is pretty slow. You can pretty much get the same throughput as just using the SPI mode.

The sync interface however is _much_ faster. But of course, needs more pin ....

What I'd suggest :
 * Make the clock input solder jumper configurable between the 12 MHz (current option) and the sync fifo clk out (60 MHz).
 *  Re-use one of the flash pin (like WP#) for the missing OE# signal.

(both with solder jumper with a default config of what's there currently).

daveshah1, 2019-04-10:
I think the sync FIFO interface is only available on FTDI port A. So a bit more shuffling of pins would be needed, quite possibly moving the programming interface to port B (or deal with the collision with the flash). This was discussed early on, but moving the programming port was rejected at the time as it would break compatability with Diamond Programmer and anything else hardcoded to port A. I'm not sure how big an issue this is in practice though.

smunaut, 2019-04-11:
Arf yeah you're right, I hadn't noticed that.
That pretty much kills this idea :/

(1) Moving the programming port isn't a good idea, that would break compatibility with other revision of the icebreaker. Unless you add an eeprom to differentiate VID so it can be handled automatically but then it's added BOM cost and require initial programming (and board space)

(2) Sharing port A would require a bit of external logic (to handle CS# and Reset# which are the two pins that can't be shared) which again, BOM cost and board space.

Unless someone sees another solution, I'd just close this.

kehribar, 2019-05-10:
Hi all,

I was the one who suggested that full PortA should be connected to the FPGA if anyone wants to use higher speed data transfers. See the twitter thread: https://twitter.com/ihsankehribar/status/948182497556357120

Keyword is '245 async FIFO'  and since it is async, implementation is much more easy. Also it can coexist with the JTAG interface. Downside is you get 1/5 of the max usb speed (~40Mbytes/sec) which is around 8Mbtyes/sec. Also, another _problem_ is that the users have to solder zero ohm jumpers to make the additional connections between FPGA and FTDI chip.

I didn't get involved in the crowdsupply campaign for _reasons_ and don't have the icebreaker hardware. Therefore, I didn't do any real work on the topic but I'm sharing the information anyway.

Best,
ihsan.

esden, 2019-05-15:
I have removed the V1.0d milestone from this. I don't think there is a good clean way of swapping the FIFO interface in the short term.

I think a high speed USB interface is something we will have to explore for V2.0 using the FTDI replacement further down the road. So I am keeping this open for now. We will have to see what we can do later.

For those wondering and checking this issue out. We did not want to put the programming on the second channel as it is usually not the default in the programmer software out there. This is an educational board and any special treatment necessary for the board is higher priority issue than one extra jumper feature for power users. We try to accommodate power users as much as we can but we will not compromise the ease of use for the beginner to do that.

closed 2020-06-25 by smunaut