icebreaker-fpga/icebreaker #28: Change the AUX CS jumper to Normally Closed

esden, 2019-05-15:
The AUX CS jumper can be closed to connect the first FTDI channel (used normally for bitstream flash or FPGA SRAM programming) to the FPGA Red LED pin. This jumper allows us to stream SPI data into the FPGA resuing the programming port after the programming is done.

One current example using this system is driving LED panels using the @smunaut  [rgb_panel design](https://github.com/smunaut/ice40-playground/tree/master/projects/rgb_panel). Also the project [SLED out_mpsse_spi module](https://github.com/shinyblink/sled/) supports this protocol and can be used as a data source.

Several of us have now used our iCEBreakers with the jumper set, and it does not seem to interfere with any normal operation of the board even if not in use. To make things easier for the users that want to quickly play around with LED panels we should change the NO jumper to NC. This will allow for separation of this connection if some issue appears down the road.

smunaut, 2019-05-15:
The only downside I could see is if some programmer software out there puts that pin as output and doesn't release it after the fpga is configured.

esden, 2019-05-15:
Agreed, I guess we will have to put it out there and see what happens. But we should probably test with the vendor programmer software ourselves to see if it is an issue with the tools we can think of or not.

closed 2019-05-16 by esden

esden, 2019-05-16:
I have tested the board in Radiant Programmer. It seems to work fine. I call it good and I have changed the jumper to normally closed.

For reference on Radiant settings I have quickly added screenshots to the wiki: https://wiki.icebreaker-fpga.com/wiki/Getting_started

commit ef6c1941 (2019-05-16): Changed the AUX CS jumper to be Normally Closed.

This closes #28