icebreaker-fpga/icebreaker #35: Consider adding suggested part numbers for suitable RGB leds to the doco

TomKeddie, 2019-08-19:
Am looking to add an RGB led but want to be sure I'm ordering something suitable.

Perhaps [150352M173300](https://www.digikey.ca/product-detail/en/wurth-electronics-inc/150352M173300/732-11999-1-ND/8557209) is suitable?  Will be happy to add this to the wiki once we have some confirmed part numbers.  Thanks.

esden, 2019-08-25:
Hey @TomKeddie,

The part that the footprint is designed for is the [CLX6F-FKC](https://www.digikey.com/products/en?keywords=CLX6F-FKC) from Cree. I have bought and used  [CLX6F-FKC-CKNNQDGBB7A363](https://www.digikey.com/short/prq27j)  before with success. I have not compared the footprints but I bet other vendors make RGB LED in compatible packaging. Let us know how it goes.

Cheers,
Piotr

TomKeddie, 2019-08-31:
Here is a photo of the cree part loaded, pin 1 is blue cathode and is oriented towards the B pin.
[image]

closed 2019-08-31 by TomKeddie