icebreaker-fpga/icebreaker #36: Need to document the user hacks.

kbob, 2019-09-02:
It would be good to add a section on hardware hacking. Here's what I know of.

The WS2812 ear requires two 10K resistors in 0603 packages and a BSS138 MOSFET in SOT23 package. And a header.

The RGB LED pad fits a Cree CLX6F LED.

You can piggyback a QSPI RAM chip onto the flash chip and bring the CS pin out to the nearby pad. The suggested part is Lyontek LY68L6400. It is available from lcsc.com. The datasheet is here. http://esp32.de/lyontek/pSRAM/LY68L6400_0.3.pdf

esden, 2019-10-24:
Thanks, the pull-up value for the WS2812 ear should probably be 1K as described in #37 :)

smunaut, 2020-05-24:
Maybe create a wiki page here and then add a link to it in the readme ?