icebreaker-fpga/icebreaker #38: Frequency accuracy of the 12 MHz oscillator (U7)

andrsmllr, 2019-12-08:
What is the frequency accuracy of the 12 MHz oscillator (U7) on the Icebreaker board?
The schematic does not define the part number, is a BOM available somewhere?

andrsmllr, 2019-12-08:
OK, so the part name is in fact available in the Kicad project (of course). But the information is still not sufficient to determine the oscillators ppm rating, since the exact part name is not given.
The part is a SiT2001, which is listed here https://www.sitime.com/products/mhz-oscillators#cat226.
The complete part number should look like this "SiT2001BI-**S2**-18E - 25.000025D" and the Sx part denotes the frequency stability (1=+-20ppm, 2=+-25ppm, 3=+-50ppm).
I could not find the complete part number anywhere in the repository, my question therefor remains the same.

esden, 2019-12-08:
Hi @andrsmllr and welcome to the iCEBreaker community. :)

Thank you for your question. The oscillator that we are mounting on the iCEBreaker is the SIT2001BI-S2-33E-12.000000G. This means it is the +-25ppm version.

I hope this helps.

Cheers,
Piotr

closed 2019-12-08 by esden