icebreaker-fpga/icebreaker #39: Seamless integration of ICEbreaker to Arduino IDE (using soft core)

Harvie, 2020-01-25:
Hello. I saw several projects aiming to integrate softcore to Arduino IDE in a way that the FPGA can then be programmed in C quite easily without much user setup.

For example the lattuino project:
http://fpgalibre.sourceforge.net/Lattuino_en/index.html#tp2
https://github.com/INTI-CMNB/Lattuino_IP_Core
https://github.com/INTI-CMNB/AVR_for_iCE40_IP_Core

Or the FPGArduino:
http://www.nxlab.fer.hr/fpgarduino/
https://github.com/f32c/f32c

Do you think, you can provide such support for icebreaker board? (doesn't need to be AVR, might as well be RISC-V)

I fully understand the difference between programming FPGA in HDL and programming softcore in C. But i think such integration would be great showcase for beginners who come from arduino world to slowly ease into FPGA...

Thanks for considering this.

mithro, 2020-01-25:
If using C (or Rust, or MicroPython) on an iCE40 FPGA is attractive to you, you might want to checkout the Fomu workshop @ https://workshop.fomu.im - a lot of the work there could be easily ported to the icebreaker as they have the same FPGA too.

esden, 2020-01-27:
@Harvie thank you for asking. This is indeed on my radar. Having RISC-V Arduino is definitely something I would like to see for the iCEBreaker.

I did ask people would like to see as an example repository for the iCEBreaker:
* Twitter: https://twitter.com/esden/status/1221873800507080704?s=20
* Discord: https://chaos.social/@esden/103556867287942113

I think the first thing to do is to have some C/C++ Make examples available. But essentially all other environments are somewhat related and can be derived from that. Especially if we build the register files from SVD descriptions.

That said I will eventually get to Arduino. But if you want it faster and want to create an example repository yourself I am very happy to take a look at it and add it to the organization.

Cheers,
Piotr

drtrigon, 2021-11-08:
There is also: https://github.com/drtrigon/fpgarduino-icestorm

It is actually working already and does it's job quite well. I built an Arduino project using it: https://github.com/drtrigon/sketchbook/tree/master/Alhambra-picorv32_AirQuality

The point where I stopped development was while setting up `micros()` as I had issues getting delays on the microsecond level, all I was able to accomplish was about 10-20us. It seems to be related to the memory used on the Alhambra 2 board. The solution I came up with was to add a PLL and speed-up from the clock of 12MHz. But I am not sure whether that would help and also I guess I would need to have a clock of about 120MHz and I doubt that will work - but I did not test it yet!
(for more details see e.g. https://github.com/cliffordwolf/picorv32/issues/126)