icebreaker-fpga/icebreaker #40: "J20 Flash" header - add single-row PMOD option?

fm4dd, 2020-01-26:
Would it be possible to re-arrange the "J20 Flash" header? Shift IO0-3 pins two positions right, add a 3.3 pad next to SCK, and move GND left would give one more single-row PMOD option. I haven't fully checked the board design yet, here is a quick picture of what I mean:

[image]

fm4dd, 2020-01-27:
Never mind, I looked at the schematics. The pins are FTDI IO and not FPGA. All FPGA pins are already brought out.

closed 2020-01-27 by fm4dd

esden, 2020-01-27:
You are partly correct though. These IO pins go to the FLASH and after the FPGA is configured they could theoretically be used to talk to another device. You just have to be "careful" what you are connecting and how.

The actual issue that I would have with changing the pin order is backwards compatibility. There are plans to make hats for the iCEBreaker that connect to different pins to implement a variety of different features using more than just two Pmod. Changing the position and layout of the pins is pretty much out of the question unless it can be left backwards compatible to the current layout.