icebreaker-fpga/icebreaker #42: Too much load on the SRAM slave select line on the bitsy

vr2045, 2020-04-08:
This is wrt the bitsy:

LEDG is hooked up to the SRAM slave select. The FPGA IO can source 4-8 mA at most. 

If I read the schematic right, LEDG is hooked up to this same signal that has an LED & 330 ohm resistor which will take up about 8mA to run.

This may work at slow speeds but will potentially have very slow fall time when selecting the SRAM.

You may want to consider increasing the LED series resistor from 330 ohms to 1K or so to allow sufficient drive from the FPGA.

esden, 2020-04-08:
I think you are looking at it wrong. The LED is hooked up to 3v3 not to GND. Also it is just the CS that does not have such high speed requirements as the other pins.

vr2045, 2020-04-08:
Quite possible I misunderstood the schematic.

In my understanding, when the CS to the SRAM is pulled low to activate the SRAM, this will light up the LED and also enable the SRAM at the same time, right? Will consume about 8mA through the LED reducing edge rate of the signal considerably.

When disabling the SRAM, you dont have a problem as the LED+resistor will be the strong pullup.

smunaut, 2020-05-24:
So :

(1) With the led Vf of 2V, the current is only about 4 mA
(2) The UP5k IO are rated for 8 mA _min_. Experiments show they'll drive twice that with no issues whatsoever
(3) I've driven the PSRAM at 147 MHz ... can't go much faster than that ... and the edge rate on CS on the scope is no different than any other IOs.

Closing since both the theory and reality agree that this works just fine.

closed 2020-05-24 by smunaut