icebreaker-fpga/icebreaker #61: Board file or EasyEDA footprint

SparkyNZ, 2025-07-21:
I cannot seem to find a .brd file for the PCB layout of the iCEBreaker. What I'm hoping to do is making a daughter board that I can attach the iCEBreaker to so it will connect to my existing chain of PCB modules (like a daughter/motherboard for the iCEBreaker to plug into).

Would it be possible to add a .brd file or upload a footprint to EasyEDA please?

I can probably figure one out myself once my iCEBreaker arrives - just trying to be proactive and perhaps an EasyEDA footprint could benefit the community.

dragonmux, 2025-07-21:
You might need to use an older version of KiCad to start with if trying to load older versions of the board, but the board files are KiCad files. The current v1.1a hardware (which is what you'll get) is in modern KiCad format though and can be opened with KiCad 9 just fine.

SparkyNZ, 2025-07-22:
Thanks for your help. I'll close the issue.

closed 2025-07-22 by SparkyNZ